arduino/nicla-sense-me-fw #40: Unable to get correct quaternion value

prabodhprakash, 2021-11-13:
### Unable to get correct quaternion value

#### Nicla Sense ME

#### Arduino IDE 1.8

#### What version of Mbed-os are you using (tag or sha) ?

#### Arduino IDE 1.8 and Nicla Sense ME

#### Simply go to File-->Examples-->Arduino_BHY2-->Standalone and modify the code to show quaternion value. The values should be float and from 0-1 as mentioned in the documents here: https://docs.arduino.cc/tutorials/nicla-sense-me/getting-started#hardware--software-needed

However, I get some random integer value like below
```
12:28:50.693 -> 15420 120 -5402 -1196
12:28:50.693 -> 15420 120 -5402 -1196
12:28:50.729 -> 15420 120 -5402 -1196
12:28:50.729 -> 15420 120 -5402 -1196
12:28:50.729 -> 15420 120 -5402 -1196
12:28:50.766 -> 15421 119 -5401 -1196
12:28:50.766 -> 15421 119 -5401 -1196
12:28:50.766 -> 15421 119 -5401 -1196
```
My code is as below

```
#include "Arduino.h"
#include "Arduino_BHY2.h"

SensorQuaternion quaternion(SENSOR_ID_RV);

void setup()
{
  Serial.begin(115200);

  BHY2.begin();
  quaternion.begin();
}

void loop()
{
  BHY2.update();

  Serial.print(quaternion.w());
  Serial.print(" ");
  Serial.print(quaternion.x());
  Serial.print(" ");
  Serial.print(quaternion.y());
  Serial.print(" ");
  Serial.println(quaternion.z());

  delay(10);
}
```

prabodhprakash, 2021-11-16:
@per1234 is there any update on this?

giulcioffi, 2021-11-16:
Hi @prabodhprakash, yes there was an error in quaternion data parsing in the BHY2 libraries. Please have a look at PR #41 which should solve the issue.

closed 2021-11-18 by giulcioffi