arduino/nicla-sense-me-fw #42: (Linked) Bug report: Nicla is disconnecting/crashing after ~30 min of I2C com

lsi8, 2021-12-07:
### Description of defect

I first reported [this](https://github.com/arduino-libraries/Arduino_Pro_Tutorials/issues/21#issue-1073045588) bug at [arduino-libraries/Arduino_Pro_Tutorials](https://github.com/arduino-libraries/Arduino_Pro_Tutorials) but I was told by @per1234 that it should also be tracked here.

@marqdevx 
I am trying to get your basic [tutorial](https://docs.arduino.cc/tutorials/nicla-sense-me/use-as-mkr-shield#troubleshooting) working for the Nicla and Portenta H7 connected over ESLOV. The bug in your example is that it only works for a little over half an hour before crashing.

I modified your code (technically, @giulcioffi & @polldo 's code) for debugging purposes:
**[nicla-sense-me-fw/Arduino_BHY2Host/examples/Accelerometer/Accelerometer_copy.ino](https://github.com/arduino/nicla-sense-me-fw/blob/e8956b952ad1734745cbc2cf530e9c3ce1777aa4/Arduino_BHY2Host/examples/Accelerometer/Accelerometer.ino):**
```cpp
#include "Arduino.h"
#include "Arduino_BHY2Host.h"
#include <LandoRGBLedPortenta.h>

SensorXYZ accel(SENSOR_ID_ACC);
LandoRGBLedPortenta portentaLeds;

const int MAX_DUPLICATE_READINGS = 10;
String colorToToggle = "white";
int duplicateReadings = 0;

static unsigned long printTime = 0;
static unsigned long startTime = 0;
static String timeToCrash = "";
static String lastAccelReading = "";

void setup()
{
  Serial.begin(115200);
  while(!Serial);
  Serial.println("Serial started!");
  portentaLeds.setColor("red");
  BHY2Host.begin(false, NICLA_VIA_ESLOV);
  Serial.println("BHY2HostSuccess!");
  portentaLeds.setColor("green");
  accel.begin();
  Serial.println("Leaving setup!");
  portentaLeds.setColor("blue");
  
  printTime = millis();
  startTime = printTime;
}

void loop()
{
  BHY2Host.update();

  if (millis() - printTime >= 1000) {
    printTime = millis();
    String accelReading = accel.toString();
    if(accelReading == lastAccelReading) {
      duplicateReadings++;
      if(duplicateReadings == MAX_DUPLICATE_READINGS) {
        timeToCrash = " " + String(printTime - startTime) + " ms";
        colorToToggle = "red";
      }
    }
    else {
      duplicateReadings = 0;
    }
    Serial.println(String("Acceleration values: ") + String(accelReading) + String(timeToCrash));
    portentaLeds.toggleColor(colorToToggle);
    lastAccelReading = accelReading;
  }
}
```

**[nicla-sense-me-fw/Arduino_BHY2/examples/App/App_copy.ino](https://github.com/arduino/nicla-sense-me-fw/blob/e8956b952ad1734745cbc2cf530e9c3ce1777aa4/Arduino_BHY2/examples/App/App.ino):**
```cpp
#include "Nicla_System.h"
#include "Arduino.h"
#include "Arduino_BHY2.h"

void colorCycle(int loops) {
  for(int i = 0; i < loops; i++) {
    nicla::leds.setColor(red);
    delay(1000);
    nicla::leds.setColor(green);
    delay(1000);
    nicla::leds.setColor(blue);
    delay(1000);
  }
  nicla::leds.setColor(off);
}

void setup(){
  BHY2.begin(NICLA_I2C, NICLA_VIA_ESLOV);
  nicla::leds.begin();
  colorCycle(5);
}

void loop(){
  // Update and then sleep
  nicla::leds.setColor(blue);
  BHY2.update(500);
  nicla::leds.setColor(off);
}
```

After 1,881,243–2,398,485 ms, the Nicla stops updating…
* The blue led on Nicla stops flashing indicating that the `BHY2.update(500);` line in the loop probably caused an exception.

[image]

If I reset just the Portenta and relaunch the serial printer, the Portenta halts at the `accel.begin();` line in the setup...

[image]

#### Target(s) affected by this defect ?
Refer to bug description.

#### Toolchain(s) (name and version) displaying this defect ?
Refer to code.

#### What version of Mbed-os are you using (tag or sha) ?
Most recent versions required by the linked code as of today (12/7/2021).

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Most recent versions required by the linked code as of today (12/7/2021).

#### How is this defect reproduced ?
By running the linked code and waiting over 30 min.

marqdevx, 2021-12-07:
Hello @lsi8
I've tried your sketches, without using your `<LandoRGBLedPortenta.h>` library

And it has been working for 2 hours with no crashes.

My suggestions is when you plug the Nicla into the Portenta, tap the reset button of the Nicla once, before openning the Serial monitor.

lsi8, 2021-12-07:
> Hello @lsi8 I've tried your sketches, without using your `<LandoRGBLedPortenta.h>` library
> 
> And it has been working for 2 hours with no crashes.
> 
> My suggestions is when you plug the Nicla into the Portenta, tap the reset button of the Nicla once, before openning the Serial monitor.

I’ve tried that sir, I’ve tried every combination over the past few weeks and nothing works for me. Can you post the most recent print outs from you serial printer?

marqdevx, 2021-12-07:
> I’ve tried that sir, I’ve tried every combination over the past few weeks and nothing works for me. Can you post the most recent print outs from you serial printer?

I have the values, and they are changing if I rotate the Nicla.

Could you please check that you have the latest versions of the Portenta and Nicla Cores, also check the Arduino_BHY2 and Arduino_BHY2Host libraries. Send me which versions you are using.

lsi8, 2021-12-07:
> Could you please check that you have the latest versions of the Portenta and Nicla Cores, also check the Arduino_BHY2 and Arduino_BHY2Host libraries. Send me which versions you are using.

I was running 2.5.2 for portenta core but 2.6.1 is available (installing now).

I have the latest nicla core:
[image]

I have the latest libraries:
[image]

lsi8, 2021-12-07:
@marqdevx 
Okay I am running this now after updating the installed Portenta board to 2.6.1. Will let you know if I make it past an hour (40 min is the record from yesterday).

lsi8, 2021-12-07:
@marqdevx
This is somewhat unrelated—however could be tested using the same linked code with the basic [modifications](https://forum.arduino.cc/t/change-the-sample-rate-for-arduino-nicla-sense-me/931824/8) for speed increase—but how fast have you managed to get this data over I2C?

10 Hz is the fastest I can get the data without communication breaking. I was hoping to get, not only the acceleration data, but data from all other sensors at over 100 Hz.

lsi8, 2021-12-07:
@marqdevx 
It stopped working after 927,549 ms.
[image]

It is worth noting that hl-zhang00 at the previously linked Arduino forum [thread](https://forum.arduino.cc/t/change-the-sample-rate-for-arduino-nicla-sense-me/931824/8) reported the same behavior as me.

marqdevx, 2021-12-10:
But that only says that you had the same value few times, if you rotate the board, the data is not changing?

lsi8, 2021-12-10:
> But that only says that you had the same value few times, if you rotate the board, the data is not changing?

I continue running the loop after I detect 10 repeats. Even hours after the detection I am still receiving the exact same values. This isn't physically possible. No the data does not change when I rotate or move the board after this event occurs.

marqdevx, 2021-12-10:
> I continue running the loop after I detect 10 repeats. Even hours after the detection I am still receiving the exact same values. This isn't physically possible. No the data does not change when I [...]

Hmm could you remove the `LandoRGBLedPortenta` library and try again?

I will try again, and if mine is working I will check if the issue is on the firmware of the Board.

I will get back to you on Monday!

Have a nice weekend.

lsi8, 2021-12-10:
I was having the issue before I included that library. I was having the issue with the barebones code from your tutorial. I will still try it later today and share the results though. Thanks for your help.

giulcioffi, 2021-12-13:
Hello @lsi8, I let your sketches run on my boards for around 3 hours and never encountered your issue. I didn't include the `LandoRGBLedPortenta` library, but that shouldn't change the behavior.
Anyway, I noticed that you were concerned about the frequency at which the sensors are updated. You could actually set the sample rate (`Hz`) and the latency (`ms`) from the `begin()` function. For instance, `accel.begin(1000,1)` means that you are setting a sample rate of 1kHz and a latency of 1ms for the `accel` sensor.
However, thanks to your issue, I noticed that there was a bug in the code which was limiting the maximum "working" sample rate. In particular, in the `Arduino_BHY2Host` library, I was reading the available sensors inside a while loop. This loop had as condition the number of available sensor data. In this condition, if you set a sample rate higher than the frequency at which you actually read the data, you will get stuck in that while() loop and never be able to reach the line in the sketch that prints the sensor values.
I pushed a [commit](https://github.com/arduino/nicla-sense-me-fw/commit/092ebe2a84f1a7d4a8930592183918a6fed9786f) that solves this issue.
I also updated the default values for sample rate and latency in this [commit](https://github.com/arduino/nicla-sense-me-fw/commit/89390f7ae2ca7a17919a8aa2a1e238ecce0c481a). Now they are set to 1kHz and 1ms by default.
Could you please run some tests with these latest changes and let me know if you still encounter problems? Thanks!

Please note that these fixes are currently available only in the main branch of this repository (and the mirrored ones). They have not been released yet, so please replace your downloaded libraries with the source code fetched from here.

xamix, 2022-03-31:
Same problem here, with the default sensor sketch.
The board seem to `stop working after sometimes (on last try it stopped after 12minutes)
I'm on 3.0.1 Arduino OS Nicla Bards
And on 1.0.4 Arduino_BPHY2

Also I tried to add SensorActivity sensor and the program crash if you use orientation sensor before:

```
/* 
 * This sketch shows how nicla can be used in standalone mode.
 * Without the need for an host, nicla can run sketches that 
 * are able to configure the bhi sensors and are able to read all 
 * the bhi sensors data.
*/

#include "Arduino.h"
#include "Arduino_BHY2.h"

// SensorXYZ accel(SENSOR_ID_ACC);
// SensorXYZ gyro(SENSOR_ID_GYRO);
// Sensor temp(SENSOR_ID_TEMP);
// Sensor gas(SENSOR_ID_GAS);
// SensorQuaternion rotation(SENSOR_ID_RV);
SensorOrientation orientation(SENSOR_ID_DEVICE_ORI);
SensorActivity activity(SENSOR_ID_AR);

void setup()
{
  Serial.begin(115200);
  while(!Serial);

  BHY2.begin();

  // accel.begin();
  // gyro.begin();
  // temp.begin();
  // gas.begin();
  // rotation.begin();
  orientation.begin();
  activity.begin();
}

void loop()
{
  static auto printTime = millis();

  // Update function should be continuously polled
  BHY2.update();

  if (millis() - printTime >= 1000) {
    printTime = millis();

    // Serial.print(String("acceleration: ") + accel.toString());
    // Serial.print(String("gyroscope: ") + gyro.toString());
    // Serial.println(String("temperature: ") + String(temp.value(),3));
    // Serial.println(String("gas: ") + String(gas.value(),3));
    // Serial.print(String("rotation: ") + rotation.toString());
    Serial.println(String("orientation: ") + orientation.toString());
    Serial.println(String("activity: ") + activity.getActivity());
    Serial.println("");
  }
}
```

Here is the output, and as you can see, the Activity sensor seem to output the Orientation values instead of the activity values:

```
orientation: Orientation values - heading: 1.000   pitch: 2941.000   roll: 1.000

activity: Orientation values - heading: 1.000   pitch: 2941.000   roll: 1.000

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R   0: AF3704EC
R   1: 000010D8
R   2: 8F3614EC
R   3: 2000F000
R   4: 2000DF30
R   5: 2000DF28
R   6: 200004E8
R   7: 20000A7C
R   8: 00000000
R   9: 00000000
R  10: 00000000
R  11: 00000000
R  12: 000010D9
SP   : 20003D78
LR   : 0001983B
PC   : 0004E85E
xPSR : 010F0000
PSP  : 20003D10
MSP  : 2000FFC0
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000082
UFSR : 00000000
DFSR : 00000000
AFSR : 00000000
BFAR : AF3704F0
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
```

Any hint how to debug that?

giulcioffi, 2022-05-02:
Hi @xamix, thanks for reporting the issue with the activity sensor. It was related to a missing return value and should be now fixed by https://github.com/arduino/nicla-sense-me-fw/commit/f04953dc3e3248878026ae4eab43ea7277e3b00f.
About the board that seems to stop working, are you still talking about the Standalone sketch? If so, this may be fixed by https://github.com/arduino/nicla-sense-me-fw/commit/188161378bb9dfce599a2e25d19a1c979212d003.
Would you mind testing with the source code from the main branch and letting me know if the issue is fixed or not? Thanks!

sebromero, 2022-09-08:
Hi @xamix ! Were you able to test it again?

xamix, 2022-11-22:
Hi,

Sorry for the delay, I just get time to test back.
I tested with the 1.0.5:

The problem with activity Sensor is indeed FIXED
For the problem with sensor, I do not have tested enougth for now but I haven't got any hang for a while with 1.0.5

sebromero, 2022-11-22:
@xamix No worries, thank you for testing it. Looks like things are working now. Please keep us posted if you still experience issues.

MuWind-Seba, 2024-09-14:
Hello,
I am facing the same problem. Everything was running well and suddenly, the program stops (without error message or warning) at the sensor begin() function. I tried resetting, reinstalling everything, different computer, etc. No success.

I tried again with the "temperature.ino" sketch found on github: https://github.com/arduino-libraries/Arduino_BHY2Host/blob/main/examples/Temperature/Temperature.ino