arduino/nicla-sense-me-fw #43: BHY Firmware update example appears to write to the incorrect flash

nuhash, 2021-12-12:
### Description of defect
BHY Firmware update example appears to write to the incorrect flash

#### Target(s) affected by this defect ?
Nicla Sense ME - BHYFirmwareUpdate example

#### Toolchain(s) (name and version) displaying this defect ?
n/a

#### What version of Mbed-os are you using (tag or sha) ?

n/a

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
n/a

#### How is this defect reproduced ?
I read the documentation and the schematics. The documentation for the BHI260AP states that the flash connected to QSPI_D0-D3 is for XIP (and can be used for logging) and it also establishes protocols for erasing and writing to that flash. The schematics for the Nicla Sense Me show that the XIP flash is not connected to the ANNA B112 but that a flash chip is shared between the B112 and the BHI260AP (on SPI0).

The BHYFirmwareUpdate example writes firmware to the shared flash chip and not the XIP flash chip which suggests that it's not doing what it says it's doing.

I am waiting on my Nicla to arrive so I can confirm but since there is a function in the BHY2 library to update the XIP flash it seems to me that the example is incorrect.

Can someone clarify why that example is writing to the shared flash instead of the dedicated flash for the BHI260AP? If the example is actually wrong, I am happy to put together an example that is correct.

giulcioffi, 2021-12-13:
Hi @nuhash. Yes, the BHYFirmwareUpdate example writes the firmware to the shared flash chip, but then at the next boot that firmware is loaded into the XIP flash. This step is performed [inside the bootloader](https://github.com/arduino/nicla-sense-me-fw/blob/89390f7ae2ca7a17919a8aa2a1e238ecce0c481a/bootloader/src/bhy_upload.cpp#L271-L290).
I hope that this clarifies to you the firmware update process.

nuhash, 2021-12-13:
I see, I didn't think to check the bootloader.

Thanks, that makes it much clearer what's going on.

closed 2021-12-13 by nuhash