arduino/nicla-sense-me-fw #48: Arduino_BHY2 sample rate

tk-king, 2022-01-23:
Hi, 
I tried to use the Nicla to record data from various sensors at the same time.
When selecting two sensors (SENSOR_ID_ACC and SENSOR_ID_GYRO) at the same time, with a sample rate of 50 Hz and a latency of 0 over WebBLE, I experience that data gets dropped.

What is the maximum sample rate achievable with the Nicla when using a single or multiple sensors?

Kind regards.

giulcioffi, 2022-01-26:
Hi @tk-king, each sensor can reach different sample rates. For example temperature sensor works with very low sample rate, while others, such as the accelerometer, can reach higher ones.
If it is not a "slow" sensor, the BHY chip provides data at the sampling rate that you set and writes them into a queue. This queue is handled by our library and data are fetched from this queue as soon as someone requests them.
If the data request is slower than the frequency at which data are generated, some data are dropped.

I tried to measure the actual sample rate with the WebBLE app. I enabled just one sensor (`SENSOR_ID_ACC`) and set `sample rate = 50 Hz` and `latency = 20 ms`. I measured that the WebBLE app was able to collect around 50 sensor data per second. Which is in line with the parameters that I set.
Then if I try to enable both sensors (`SENSOR_ID_ACC` and `SENSOR_ID_GYRO`) with the same parameters, I can collect only around 27 samples per sensor. Which makes a total of 54 data per second.
Please notice that if on Nicla Sense you are running the sandard App.ino sketch, you are introducing a delay of 100ms between the `BHY2.update()` calls. If you remove that delay, you can reach around 70 data samples per second.

70 Hz seems to be the maximum frequency at which a single sensor can be sampled with the WebBLE app without dropping data. The bottleneck is then the overall communication time.

gumush, 2022-04-10:
Actually these motion sensors can provide over 1000hz. Is there any way to store them in Flash while capturing.After capturing it can be transmitted over bluetooth ?

bstbud, 2023-06-03:
@gumush 
to stream very high rates, it's possible.
If using USB, you could stream data of 1600hz of accel and gyro simultaneously using example provided in this PR.

when using BLE to stream the data, 
I found this [script ](https://github.com/lucasguocn/Adafruit_CircuitPython_BLE/blob/nicla_main/test_nicla_sensors.py) useful which runs on a Raspberry PI or Linux host. This script uses batching of sensor packets by setting the "latency": to a non-zero value. when the sample rate is 100hz, a latency of 90ms essentially means that you have 10 samples in each batch, this greatly increases the throughput. Based on my quick testing, sampling of 800hz could be achieved so far.

closed 2023-06-03 by per1234