arduino/nicla-sense-me-fw #49: What does accelerometer/gyroscope/magnetometer data mean?

hkayann, 2022-02-18:
Is there any documentation that shows what IMU data means? For example, accelerometer is not in g's. It is not m/s^2 as well. 

Also I need to set the offset as well, as at standing still position, I do not get 0's.
[image]

hkayann, 2022-02-18:
I feel like, I am definitely doing something wrong. This is the gyroscope reading which is very noisy.

[image]

I would expect a similar output, that I get from AdaFruit Feather Sense as shown below, which looks well-calibrated.

[image]

hkayann, 2022-02-24:
Here is the example data generated by the example standalone script:

```
acceleration: XYZ values - X: -523   Y: 437   Z: -4022

gyroscope: XYZ values - X: 0   Y: 0   Z: 1

acceleration: XYZ values - X: -515   Y: 438   Z: -4026

gyroscope: XYZ values - X: -2   Y: -2   Z: 0

acceleration: XYZ values - X: -516   Y: 444   Z: -4019

gyroscope: XYZ values - X: 0   Y: -2   Z: 0

acceleration: XYZ values - X: -510   Y: 448   Z: -4028

gyroscope: XYZ values - X: -1   Y: 0   Z: 0

```
What are the units of these data?

facchinm, 2022-02-25:
@bstbud can you provide some insight on the data format? Thanks!

bstbud, 2022-02-28:
@facchinm 

The table 88 (~page 119) in [BHI260AP datasheet]([url](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bhi260ap-ds000.pdf) describes the data format of all sensors.
Basically some sensors have fixed scale factors while the others have dynamic scale factors such as for accelerometers and gyroscopes.

The built-in sensor HW such as accelerometer/gyroscope has dynamic ranges that could be configured.
And this dynamic range determines the scale factor of the sensor.
section: 13.2.8 Change Sensor Dynamic Range (0x000E) of the BHI260AP datasheet has more details about the dynamic range.

For example, the accelerometer has a default range of (+/-8G) for 16bit ADC resolution (65536LSB), thus the scale factor is: 65536 LSB / 16g,  when means 1g is 4096LSB, when the device is stationary and roughly flat, the Z axis should get a value of 4096/1g or -4096/-1g (depending on the orientation). And from the data above it's exactly what is expected.

To make it programmatically easier, there is API to retrieve the actual scale factor for sensors with dynamic range/scale factors. And this is actually demonstrated in the PR #47: https://github.com/arduino/nicla-sense-me-fw/pull/47

hkayann, 2022-02-28:
Thanks @bstbud. I converted it to m/s-2 now. I appreciate if you can confirm the following as well.

For the **gyro** I need to do **gyro / 32.768** so I can get **dps.**

In the [BMM150 datasheet](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bmm150-ds001.pdf), they describe as:

> The API/driver performs an off-chip temperature compensation and outputs x/y/z magnetic field data in 16 LSB/µT to the upper application layer: 

But, the range differs for x&y and z.

[image]

So, If I would like to get **uT per second**, I should do the following:

- x = x / (65536 / 1300), // same for y as well
- z = z / (65536 / 2500)

bstbud, 2022-03-02:
@hkayann, 

updated, mag data needs to be divided by 16.384 to be precise (previously I said 16)

hkayann, 2022-03-02:
@bstbud if we look at those ranges, shouldn't be (65536 / 2600) for x and y, and (65536 / 5000) for z?

bstbud, 2022-03-03:
Hi @hkayann 
The mag sensor data is read out by the BHI260AP chip and presented to the clients / users of the BHI260AP in a format/scale different from what you get when you directly has access the BMM150 chip. Let me know if this makes sense.
so again, to confirm, the mag data you get from BHI260AP has a (up) scale of 16.384.

hkayann, 2022-03-03:
Oh, I see. Many thanks.

closed 2022-03-03 by hkayann

bstbud, 2022-03-22:
@hkayann, it turns out the data I got previously was up to correction, and please see updated info on the scaling for the mag sensor above

hkayann, 2022-03-22:
@bstbud, so it is (65536/2600) for x&y, (65536/5000) for z. Now it makes sense for me as well.

I was not able to exactly grasp the idea of treating them same while they have different ranges.

Thanks.