arduino/nicla-sense-me-fw #55: Bug with Serial + Hardware Serial when using different baud rates

CiprianFlorin-Ifrim, 2022-04-17:
### Description of defect

#### Target(s) affected by this defect ?
Nicla Sense ME

#### Toolchain(s) (name and version) displaying this defect ?
Mbed OS Nicla Boards 3.0.1
Latest bootloader
Arduino 1.8.16

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed OS Nicla Boards 3.0.1
Latest bootloader available in the Nicla FW repo
Arduino 1.8.16

#### How is this defect reproduced ?
Simple example code:
```
#include <Arduino.h>
#include "Nicla_System.h"

void setup()
{
  nicla::begin();
  Serial.begin(115200);
  Serial1.begin(9600);

  delay(1000);
}

void loop() {
  while(Serial.available()) {
    char incomingCharacter = Serial.read();
    if (incomingCharacter == '1') {
      Serial.println("Command activated");
    }
    delay(100);                   
  }
}
```

facchinm, 2022-04-19:
Hi @CiprianFlorin-Ifrim ,
in Nicla Sense `Serial` and `Serial1` are indeed the same object (and the same couple of pins) due to how serial ports are defined in the core (see here https://github.com/arduino/ArduinoCore-mbed/blob/master/cores/arduino/Arduino.h#L108-L110 ).
Moreover, we can't instantiate two different serials since the onboard nRF52832 only has one hardware UART instance.

Maybe it would be less confusing for the end user to undefine `Serial1` object if only one UART is available. @per1234 @sebromero  what do you think about that?

closed 2022-04-19 by facchinm

per1234, 2022-04-19:
>Maybe it would be less confusing for the end user to undefine Serial1 object if only one UART is available.

Yes. I said the same back during the discussion that somehow resulted in that code being added:

https://github.com/arduino/ArduinoCore-mbed/pull/54#issuecomment-691636008

CiprianFlorin-Ifrim, 2022-04-20:
Thank you for the replies. I was not aware it was the same hardware serial and spent a few good hours on my main code thinking I had some issues there. 

I think it would be good to add in the Cheat Sheet that they are the same and same baur rate should be set for both: https://docs.arduino.cc/tutorials/nicla-sense-me/cheat-sheet#uart

sebromero, 2022-04-27:
> Maybe it would be less confusing for the end user to undefine Serial1 object if only one UART is available. @per1234 @sebromero what do you think about that?

@per1234 @facchinm  Agree, I think it would be less confusing. Let's do it 👍