arduino/nicla-sense-me-fw #75: Monitoring motion or pickup sensors has no value change

creadicted, 2023-01-14:
### Description of defect

When monitoring the sensor values for : 
57 | Wake gesture | SENSOR_ID_WAKE_GESTURE | Sensor 
59 | Glance gesture | SENSOR_ID_GLANCE_GESTURE | Sensor
61 | Pickup gesture | SENSOR_ID_PICKUP_GESTURE | Sensor

No movement of the sensor leads to value changes.

bstbud, 2023-01-19:
Hello, can you explain a bit more about the problem?
do you mean:
when the sensor reports an event even when there is no motion at all, 
or do you mean:
when the sensor is on, you can never trigger an event using whatever motion?

bstbud, 2023-01-19:
Please note that those 3 sensors are so called one-shot sensors and the behavior of the sensors follow the spec defined in Android virtual sensors, and once detected they will be deactivated automatically internally. Please check the definitions of the 3 sensors here:
https://source.android.com/docs/core/interaction/sensors/sensor-types
https://source.android.com/docs/core/interaction/sensors/report-modes#one-shot

creadicted, 2023-02-25:
> or do you mean:
> when the sensor is on, you can never trigger an event using whatever motion?

Hey! I meant when the sensor is on, it never triggered an event. Most likely, my expectations are wrong. 
Thank you for the link.

closed 2024-07-06 by creadicted