arduino/nicla-sense-me-fw #82: [AE-13] .co2_eq() and .bsec.iaq() of SensorBSEC class always returns zero

aliphys, 2023-03-16:
### Description of defect

`.co2_eq()` and `.bsec.iaq()` of SensorBSEC class always returns zero

[image]

#### Target(s) affected by this defect ?

Nicla Sense ME

#### Toolchain(s) (name and version) displaying this defect ?

Version: 2.0.4
Date: 2023-02-27T16:14:28.576Z
CLI Version: 0.31.0

Copyright © 2023 Arduino SA

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

Run this sketch, which is modified from the Standalone.ino sketch
```

/* 
 * This sketch shows how nicla can be used in standalone mode.
 * Without the need for an host, nicla can run sketches that 
 * are able to configure the bhi sensors and are able to read all 
 * the bhi sensors data.
*/

#include "Arduino_BHY2.h"

SensorXYZ accel(SENSOR_ID_ACC);
SensorXYZ gyro(SENSOR_ID_GYRO);
Sensor temp(SENSOR_ID_TEMP);
Sensor gas(SENSOR_ID_GAS);
SensorQuaternion rotation(SENSOR_ID_RV);
SensorBSEC bsec(SENSOR_ID_BSEC);

void setup()
{
  Serial.begin(115200);
  while(!Serial);

  BHY2.begin();

  accel.begin();
  gyro.begin();
  temp.begin();
  gas.begin();
  bsec.begin();
  rotation.begin();
}

void loop()
{
  static auto printTime = millis();

  // Update function should be continuously polled
  BHY2.update();

  if (millis() - printTime >= 1000) {
    printTime = millis();

    Serial.println(String("acceleration: ") + accel.toString());
    Serial.println(String("gyroscope: ") + gyro.toString());
    Serial.println(String("temperature: ") + String(temp.value(),3));
    Serial.println(String("gas: ") + String(gas.value(),3));
    Serial.println(String("co2: ") + String(bsec.co2_eq(),3));
    Serial.println(String("iaq: ") + String(bsec.iaq(),3));
    Serial.println(String("rotation: ") + rotation.toString());
  }
}

```

aliphys, 2023-03-30:
After this PR is approved, would be worth testing to see if the issue is resolved. Since both relate to the BSEC library features.
- https://github.com/arduino/nicla-sense-me-fw/pull/85

aliphys, 2023-04-13:
Suggestion on the Arduino forum:
>Open the file `/libraries/Arduino_BHY2/src/sensors/SensorTypes.h` in a text editor.
Change both following values to `30` 
`#define SENSOR_DATA_FIXED_LENGTH (30)`
`#define SENSOR_LONG_DATA_FIXED_LENGTH (30)`
Initialize the BSEC sensor in your Arduino code via the legacy ID
`SensorBSEC bsec(SENSOR_ID_BSEC_LEGACY);`

Source: https://forum.arduino.cc/t/still-nicla-sense-me-bseq-data-problems/1086863/2

aliphys, 2023-05-19:
Note: More sensorIDs also return unexpected values, and is not limited to `.co2_eq()` and `.bsec.iaq()`. Looking into documenting this.

bstbud, 2023-05-29:
Hi @aliphys 
Could you try two things in the order listed
1) upload the sketch BHYFimrwareUpdate which will make sure that the BHI260AP gets the right FW version
2) due to the asynchronous polling used below, the dumping of the BSEC values will be zero before the first BSEC sensor event gets populated to the sensor FIFO buffer. This could be addressed by simply adding a delay of 3000ms (BSEC's default modes updates data every 3seconds) to the end of the setup(), 
void setup()
{
  Serial.begin(115200);
  while(!Serial);

  BHY2.begin();

  accel.begin();
  gyro.begin();
  temp.begin();
  gas.begin();
  bsec.begin();
  rotation.begin();
  delay(3000); //this is to make sure all the sensors get their 1st sensor events.
}

aliphys, 2023-06-07:
1. Uploaded the 'BHYFirmwareUpdate` sketch as recommended. CO2 and iaq values are locked at 200112 and 221 respectively and do not change as I blow over the sensor. Unlike the gas and temperature sensor, which respond to external stimuli. 
[image]
Replicated outcome on two different Nicla Sense ME boards.
2. Addition of the ` delay(3000);` statement at the end of the `setup()` function does not alter the outcome of 1.

alrvid, 2023-06-07:
It might be because on the one hand you have this definition of co2_eq():

uint32_t co2_eq() {return _data.co2_eq;}

And on the other hand there's this: Serial.println(String(bsec.co2_eq(),3));

Which is a variety of this: String(val, decimalPlaces)

But, from the Arduino Reference: "decimalPlaces: only if val is float or double. The desired decimal places."

So, it might be that the first two values, 200112, are incorrectly displayed because co2_eq() returns a uint32_t,
but that type can't be used with decimalPlaces.

aliphys, 2023-06-07:
This line in the BSEC class looks related to what is observed in the initial problem: https://github.com/arduino/nicla-sense-me-fw/blob/b4425275859b2dea1ec22c546851106df9ac8fb5/Arduino_BHY2/src/sensors/SensorBSEC.h#L16

bstbud, 2023-06-07:
Hi @aliphys 
As @alrvid pointed out, the values of co2_eq and iaq are integers and using the float variant of the String will result in wrong fomating.

Could you try this instead:
```
    Serial.println(String("co2: ") + String(bsec.co2_eq()));
    Serial.println(String("iaq: ") + String(bsec.iaq()));
    Serial.println(String("accuracy: ") + String(bsec.accuracy()));
```
Note that the accuracy is also printed out, after the BSEC sensor is turned on, it takes it about 5 mins to calibrate and then the accuracy will increase to 1 from 0. The values of iaq and co2_eq will remain constant and could be ignored before the accuracy is 1.

aliphys, 2023-06-09:
Code modified as follows:
```
/* 
 * This sketch shows how nicla can be used in standalone mode.
 * Without the need for an host, nicla can run sketches that 
 * are able to configure the bhi sensors and are able to read all 
 * the bhi sensors data.
*/

#include "Arduino_BHY2.h"

SensorXYZ accel(SENSOR_ID_ACC);
SensorXYZ gyro(SENSOR_ID_GYRO);
Sensor temp(SENSOR_ID_TEMP);
Sensor gas(SENSOR_ID_GAS);
SensorQuaternion rotation(SENSOR_ID_RV);
SensorBSEC bsec(SENSOR_ID_BSEC);

void setup()
{
  Serial.begin(115200);
  while(!Serial);

  BHY2.begin();

  accel.begin();
  gyro.begin();
  temp.begin();
  gas.begin();
  bsec.begin();
  rotation.begin();
  delay(3000); # 3 second delay, before starting loop
}

void loop()
{
  static auto printTime = millis();

  // Update function should be continuously polled
  BHY2.update();

  if (millis() - printTime >= 1000) {
    printTime = millis();

    Serial.println(String("acceleration: ") + accel.toString());
    Serial.println(String("gyroscope: ") + gyro.toString());
    Serial.println(String("temperature: ") + String(temp.value(),3));

    # remove decimalPlaces agrument in String function
    Serial.println(String("gas: ") + String(gas.value()));
    Serial.println(String("co2: ") + String(bsec.co2_eq()));
    Serial.println(String("iaq: ") + String(bsec.iaq()));

    Serial.println(String("rotation: ") + rotation.toString());
  }
}
```

After 5 minutes this is the output
```
acceleration: XYZ values - X: 390   Y: -971   Z: -3929

gyroscope: XYZ values - X: 0   Y: -2   Z: -2

temperature: 30.430
gas: 22369.00
co2: 502
iaq: 26
rotation: Quaternion values - X: 0.000   Y: 0.000   Z: 0.000   W: 0.000   Accuracy: 0.000

```

Note that:
- The values of `bsec.co2_eq()` and `bsec.iaq()` are fixed at 500 and 25. How is the User able to identify these as incorrect values?
- The quaternion values are still zero

bstbud, 2023-06-09:
@aliphys 
as mentioned above, for the BSEC sensor, there is this accuracy value which is an indicator telling the user the fidelity of the values from BSEC output.
so it's advised to add the following:
  Serial.println(String("accuracy: ") + String(bsec.accuracy()));

The quaternion values shouldn't be zero, if yes, this means that no quaternion events have been read out from the FIFO, this for me seems like a genuine issue.

bstbud, 2023-06-14:
@aliphys, this issue is about the BSEC values, if you agree with the explanations above, I propose this issue to be closed, and for the quaternion output issue, maybe it should be tracked in another issue thread?

aliphys, 2023-06-14:
@bstbud Split the quaternion item into a new issue thread (https://github.com/arduino/nicla-sense-me-fw/issues/107). Will get back on the ` Serial.println(String("accuracy: ") + String(bsec.accuracy()));` item.

aliphys, 2023-06-15:
When `bsec.accuracy()` is not zero, then the `bsec.co2_eq()` and `bsec.iaq()` values are no longer static. Replicated reliabily on multiple Nicla Sense ME boards.

```
acceleration: XYZ values - X: -627   Y: -2413   Z: 3180

gyroscope: XYZ values - X: 100   Y: -50   Z: 263

temperature: 34.540
gas: 7937.00
accuracy: 2
co2: 650
iaq: 86
rotation: Quaternion values - X: 0.000   Y: 0.000   Z: 0.000   W: 0.000   Accuracy: 0.000
```

aliphys, 2023-06-15:
Closing issue as resolved. ✅

closed 2023-06-15 by aliphys