arduino/nicla-sense-me-fw #90: [AE-15] Commander.py script: cobs.cobs.DecodeError: not enough input bytes for length code

aliphys, 2023-04-06:
### Description of defect

Upon running the Commander.py script, the `cobs.cobs.DecodeError: not enough input bytes for length code` error is displayed.

```
PS D:\Github\aliphys\nicla-sense-me-fw\Arduino_BHY2\examples\DataHarvester\extras\Commander> & C:/Users/user/AppData/Local/Programs/Python/Python39/python.exe d:/Github/aliphys/nicla-sense-me-fw/Arduino_BHY2/examples/DataHarvester/extras/Commander/Commander.py COM23 test phys
Running...
> Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Python\Python39\lib\threading.py", line 950, in _bootstrap_inner
    self.run()
  File "C:\Users\user\AppData\Local\Programs\Python\Python39\lib\threading.py", line 888, in run
    self._target(*self._args, **self._kwargs)
  File "d:\Github\aliphys\nicla-sense-me-fw\Arduino_BHY2\examples\DataHarvester\extras\Commander\Commander.py", line 88, in receiver       
    decoded = cobs.decode(data[0:(n - 1)])
cobs.cobs.DecodeError: not enough input bytes for length code
```

https://github.com/arduino/nicla-sense-me-fw/blob/f42f36179dabc4b1f3de87d018565f24219494e7/Arduino_BHY2/examples/DataHarvester/extras/Commander/Commander.py#L88

This may related to truncated data coming from the DataHarvester.ino file?

#### Target(s) affected by this defect ?
Nicla Sense ME

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?
- Ensure all required modules are installed via pip
- Use `arduino-cli` to identify the COM port
- In the terminal, run `python [COMPORT] test phys`
- Press `r` then enter to start data aquisition

https://github.com/arduino/nicla-sense-me-fw/blob/f42f36179dabc4b1f3de87d018565f24219494e7/Arduino_BHY2/examples/DataHarvester/extras/Commander/Commander.py#L55-L62

sebromero, 2023-04-06:
@manchoz Any idea what's going wrong here?

manchoz, 2023-04-06:
Something might be changed in the PacketSerial or Python COBS library. @aliphys can try changing the `n-1` index to `n` at line 88?

Is the same issue present on MacOS or GNU/Linux?

Also check for data being sent correctly from the Nicla Sense ME.

Pezcraft, 2024-10-20:
The solution is to convert the accelerometer data to float 

DataHarvester.ino...
```
Data data { now, float(accel.x()), float(accel.y()), float(accel.z()) };
```