arduino/nicla-sense-me-fw #91: BHA260 SDK Nicla Board Configuration

OrsoEric, 2023-04-06:
I need to customize the BHA260 firmware. I installed the BHA260 SDK and I'm able to generate new firmware for said chip, convert the .fw into a .h file, and upload the BHA260 firmware to the "Nicla Sense ME" board.

The BHA260 SDK doesn't have a board configuration that maps all the drivers on the "Nicla Sense ME" board, I compiled and tested a firmware that maps just the accelerometer and the gyroscope:
[image]

[The BHYFirmwareUpdate](https://github.com/arduino/nicla-sense-me-fw/tree/main/Arduino_BHY2/examples/BHYFirmwareUpdate) Repo has a 1.1MB firmware with all the drivers of the "Nicla Sense ME" board. Where can I find the BHA260 SDK project that generates said 1.1MB firmware with the drivers for all the "Nicla Sense ME" sensors?

OrsoEric, 2023-04-06:
below the output of  [the sensod ID .ino](https://github.com/arduino-libraries/Arduino_BHY2/blob/main/examples/ShowSensorList/ShowSensorList.ino) on the two firmware, on the left the SDK firmware, on the right the Nicla FW firmware, listing the virtual driver IDs:

Here is the table in markdown format:

| Bosch_APP30_SHUTTLE_BHI260_turbo-flash | Nicla Sense ME firmware |
| --- | --- |
| 1.000 | 1.000 |
| 3.000 | 3.000 |
| 4.000 | 4.000 |
| 5.000 | 5.000 |
| 6.000 | 6.000 |
| 7.000 | 7.000 |
| 10.000 | 10.000 |
| 12.000 | 12.000 |
| 13.000 | 13.000 |
| 14.000 | 14.000 |
| 15.000 | 15.000 |
| 16.000 | 16.000 |
| 28.000 | 19.000 |
| 29.000 | 21.000 |
| 31.000 | 22.000 |
| 32.000 | 23.000 |
| 37.000 | 24.000 |
| 38.000 | 25.000 |
| 48.000 | 28.000 |
| 50.000 | 29.000 |
| 52.000 | 31.000 |
| Bosch_APP30_SHUTTLE_BHI260_turbo-flash | Nicla Sense ME firmware |
| --- | --- |
| 53.000 | 32.000 |
| 55.000 | 34.000 |
| 57.000 | 35.000 |
| 59.000 | 37.000 |
| 61.000 | 38.000 |
| 63.000 | 40.000 |
| 67.000 | 41.000 |
| 69.000 | 43.000 |
| 70.000 | 44.000 |
| 75.000 | 48.000 |
| 77.000 | 50.000 |
| 94.000 | 52.000 |
|   | 53.000 |
|   | 55.000 |
|   | 57.000 |
|   | 59.000 |
|   | 61.000 |
|   | 63.000 |
|   | 67.000 |
|   | 69.000 |
|   | 70.000 |
|   | 75.000 |
|   | 77.000 |
|   | 94.000 |
| Bosch_APP30_SHUTTLE_BHI260_turbo-flash | Nicla Sense ME firmware |
| --- | --- |
|   |112.000 |
|   |115.000 |
|   |116.000 |
|   |117.000 |
|   |127.000 |
|   |128.000 |
|   |129.000 |
|   |130.000 |
|   |131.000 |
|   |132.000 |
|   |133.000 |
|   |134.000 |
|   |135.000 |

gmacario, 2023-04-14:
@manchoz is there any chance we can obtain this piece information?

manchoz, 2023-04-14:
Hi @gmacario, being a binary blob provided by the sensor manufacturer, we need to discuss it with the original author of the firmware.

We will keep you updated.

bstbud, 2023-05-29:
@gmacario 
this repo only provides some pre-built FW for the BHI260AP sensor, for customization, maybe you could raise the request on: https://community.bosch-sensortec.com/t5/Bosch-Sensortec-Community/ct-p/bst_community