arduino/nicla-sense-me-fw #97: Over-the-air update support for Nicla Sense ME?

hkayann, 2023-04-29:
Hi, 

Is doing OTA update possible for Nicla Sense ME? If yes, I appreciate any guidance.

I could only find this [OTA guide for Arduino Portenta H7](https://docs.arduino.cc/tutorials/portenta-h7/over-the-air-update).

facchinm, 2023-05-02:
Hi @hkayann ,
the `[bhy_controller](https://github.com/arduino/nicla-sense-me-fw/tree/main/tools/bhy-controller)` tool implements DFU for Nicla Sense ME using WebBluetooth. Hope it helps you getting started!

closed 2023-05-02 by facchinm

hkayann, 2023-05-02:
@facchinm  I have tried to use that tool but no success yet. I cd into `src`, and then `./bhy`.
It returns `zsh: no such file or directory: ./bhy`. Any idea what is wrong?

facchinm, 2023-05-02:
Yup, you also need to compile the tool with `go build` (or use a precompiled version of the tool from here https://github.com/arduino/nicla-sense-me-fw/releases/tag/1.0.5)

hkayann, 2023-05-02:
Many thanks! Just one final thing @facchinm.

Can I change the uploaded sketch via DFU? For example, imagine it blinks blue LED, change that to red etc.

facchinm, 2023-05-02:
You can access the DFU page from here https://arduino.github.io/nicla-sense-me-fw/tools/bhy-controller/src/static/dfu.html and select the new binary to be flashed :slightly_smiling_face:

hkayann, 2023-05-02:
Many thanks @facchinm, I really appreciate the help!