arduino/nicla-sense-me-fw #101: Roll and pitch - are mixed up in places.

b582q9, 2023-05-17:
### Description of defect

Roll should be around Y axe and pitch around X axe but roll and pitch from SensorOrientation returns this parameters as mixed up

#### Target(s) affected by this defect ?
Arduino Nicla Sense ME

#### What version of Mbed-os are you using (tag or sha) ?
Arduino Mbed OS Nicla Boards 4.0.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Arduino_BHY2 1.0.6
Arduino IDE 2.1.0

#### How is this defect reproduced ?

#include "Arduino.h"
#include "Arduino_BHY2.h"

SensorOrientation orientation(SENSOR_ID_ORI);

void setup()
{
  Serial.begin(115200);
  while(!Serial);
  BHY2.begin();
  orientation.begin();
}

void loop()
{
  BHY2.update(1);   
  Serial.print("Heading = ");
  Serial.println(orientation.heading());
  Serial.print("Pitch = ");
  Serial.println(orientation.pitch());
  Serial.print("Roll = ");
  Serial.println(orientation.roll());
}

b582q9, 2023-05-26:
@bstbud 
It is very strange because there is no pitch, roll or yaw definitions in Android word. It is belong to aviation or robots development.
So, you should change definitions in library (rename pith and roll) or change the behaviour please.

bstbud, 2023-05-26:
Hi

on page:https://developer.android.com/guide/topics/sensors/sensors_position, it says: 
<html>
<body>

SensorEvent.values[0] | Azimuth (angle around the z-axis). | Degrees
-- | -- | --
SensorEvent.values[1] | Pitch (angle around the x-axis).
SensorEvent.values[2] | Roll (angle around the y-axis).

</body>
</html>

b582q9, 2023-05-26:
@bstbud 
Ok, you made it like it shouldn't :).
But, maybe you could make a lib version with a definitions like in aviation? It will be very useful for those who want to use nicla for avionics.

per1234, 2023-05-27:
I see the information is already specified in the documentation:

https://docs.arduino.cc/tutorials/nicla-sense-me/getting-started#sensororientation

so I'll close this issue.

>maybe you could make a lib version with a definitions like in aviation?

You are welcome to create such a library @b582q9

closed 2023-05-27 by per1234