arduino/nicla-sense-me-fw #103: Data frequency drops by half when transmitter over BLE.

hkayann, 2023-06-06:
Hi,

I am transmitting data over BLE from Nicla to my Pi. I have observed that the data frequency drops by half (from 20Hz to 10Hz)after some time as seen below.
[image]

I have tried 3 times, happened every time. What might be the reason?

aliphys, 2023-07-21:
Hello @hkayann 👋 
Can you share the minimal code/setup that replicates your issue?

hkayann, 2023-07-22:
I am using the following example: 

https://github.com/hkayann/node-red-contrib-ble-sense/blob/main/examples/niclaBLEIMUStruct/niclaBLEIMUStruct.ino

I think this might be related that magnetometer freq. being fixed to 10Hz somehow. I confirmed that no environmental issues present.