arduino/nicla-sense-me-fw #109: High RAM usage of Arduino_BHY2.

hkayann, 2023-07-25:
I appreciate if someone can explain why this library consumes so much ram.

Here is the empty sketch usage:
```
RAM:   [=         ]  12.0% (used 7720 bytes from 64288 bytes)
Flash: [=         ]   7.6% (used 40184 bytes from 527616 bytes)
```
If I include **Arduino_BHY2**:
```
RAM:   [======    ]  55.5% (used 35672 bytes from 64288 bytes)
Flash: [=====     ]  50.8% (used 267952 bytes from 527616 bytes)
```
As, I am trying to run my ML model, I am having diffuculties regarding RAM availability.

bstbud, 2023-08-16:
Hi 
@hkayann 
the Ardunio_BHY2 uses mbed's BLE stack which is the biggest consumer of the RAM.
If you don't need BLE, you could use proper parameters for BHY2.begin() to turn off the BLE, which should help reduce the RAM usage significantly. Example: 
BHY2.begin(NICLA_STANDALONE);

sebromero, 2023-10-23:
Hi @hkayann ! We're you able to configure the sketch to use less RAM as suggested by @bstbud ?

hkayann, 2024-01-10:
@sebromero I am not so still looking for a way to do this. 

@bstbud  is there any way to reduce the RAM used by the BLE library?

hkayann, 2024-01-19:
Another interesting this is the `Arduino_BHY2` requires` ArduinoBLE.h` anyway.

bstbud, 2024-01-19:
> @sebromero I am not so still looking for a way to do this.
> 
> @bstbud is there any way to reduce the RAM used by the BLE library?

based on my knowledge, the memory of the BLE stack has been optimized already, and there is only little wiggling room.

hkayann, 2024-01-19:
Thanks for the reply @bstbud, I am trying alternative ways now.

One of them is to load my variables into the RAM of BHI260AP as that chip 256KB SRAM while Nicla offers 64 KB only.

I appreciate any lead on that. I know it is not directly related to this issue, sorry for that. So if you want, feel free to close this one.

bstbud, 2024-01-19:
> Another interesting thing is that the `Arduino_BHY2` requires` ArduinoBLE.h` anyway.

@hkayann, even though the ArduinoBLE.h header is included, it does not mean the RAM will be used/claimed by the BLE stack automatically although by default this is the case. If you don't need BLE in your project, you could change the parameters of the BHY2.begin() function, for example, you could pass NICLA_I2C instead of NICLA_BLE_AND_I2C (by default) and the BLE stack won't set up the memory pool in this case.

hkayann, 2024-01-19:
@bstbud  I see. I need BLE. Any suggestion on this?

_One of them is to load my variables into the RAM of BHI260AP as that chip 256KB SRAM while Nicla offers 64 KB only._

bstbud, 2024-01-22:
> @bstbud I see. I need BLE. Any suggestion on this?
> 
> _One of them is to load my variables into the RAM of BHI260AP as that chip 256KB SRAM while Nicla offers 64 KB only._

Hi @hkayann 
yes, this is the same suggestion I would give. The BHI260AP has more memory, and sensor processing related tasks could be handled on the BHI260AP which is also much more power efficient than the main MCU. The Nordic MCU could be used for connectivity tasks. Now the SDK for BHI260AP is publicly available.  You could refer to: https://www.bosch-sensortec.com/products/smart-sensor-systems/bhi260ap/#technical for how to use the SDK.

hkayann, 2024-01-23:
Hi @bstbud,

I am playing with that right now. I would really appreciate if at least, very briefly if you can tell me the steps required to implement that for Nicla Sense ME?

bstbud, 2024-01-31:
> Hi @bstbud,
> 
> I am playing with that right now. I would really appreciate if at least, very briefly if you can tell me the steps required to implement that for Nicla Sense ME?

I would start with the SDK and the quick start guide provided to implement some simple example algorithms to understand the basics and then step up to implement your own algorithm. If you encounter specific issues, I suggest to post some questions on Bosch Sensortec[ online community](https://community.bosch-sensortec.com/t5/Bosch-Sensortec-Community/ct-p/bst_community).