arduino/nicla-sense-me-fw #123: Micropython support for Nicla Sense

dspreitz, 2024-01-02:
Do you have any plans to also offer Micropython support for Nical Sense? Nicla vision is already supported my micropython.

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

facchinm, 2024-01-09:
Nicla Sense and Nicla Vision are based on very different MCUs (nRF5283 vs STM32H747). The Sense doesn't ship with enough RAM to make it a suitable target for Micropython (it would run, but with severe limitations).

closed 2024-01-09 by facchinm

dspreitz, 2024-01-09:
Martino,Thanks for the clarification and Background Info.Appreciated.