arduino/nicla-sense-me-fw #141: `Nicla_IoT_Bridge.ino` and `Portenta_BLE_Bridge.ino` sketches fail to compile.

aliphys, 2024-05-10:
[Nicla_IoT_Bridge.ino](https://github.com/arduino/nicla-sense-me-fw/actions/runs/9029376790/job/24811607057?pr=139#step:3:1146) and [Portenta_BLE_Bridge.ino](https://github.com/arduino/nicla-sense-me-fw/actions/runs/9029376790/job/24811607057?pr=139#step:3:2099) sketches from the `Arduino_BHY2Host` library do not compile for the Portenta H7. This causes the CI to fail.

Compiler output suggests that this is related to a breaking change in the ArduinoIoTCloud library.

Possibly fixed by https://github.com/arduino-libraries/Arduino_SecureElement/commit/18fb8b9603ee689ae706a019bc36e686fdf7ee7d

aliphys, 2024-07-31:
Sketches  fail to compile via the CI, while they do compile locally. 

# [Nicla_IoT_Bridge](https://github.com/aliphys/nicla-sense-me-fw/actions/runs/10180426960/job/28158266528#step:3:1159)
```
Compiling sketch: Arduino_BHY2Host/examples/Nicla_IoT_Bridge
  In file included from /home/runner/Arduino/libraries/ArduinoIoTCloud/src/ArduinoIoTCloud.h:27:0,
                   from /home/runner/work/nicla-sense-me-fw/nicla-sense-me-fw/Arduino_BHY2Host/examples/Nicla_IoT_Bridge/thingProperties.h:1,
                   from /home/runner/work/nicla-sense-me-fw/nicla-sense-me-fw/Arduino_BHY2Host/examples/Nicla_IoT_Bridge/Nicla_IoT_Bridge.ino:14:
  /home/runner/Arduino/libraries/Arduino_ConnectionHandler/src/Arduino_ConnectionHandler.h:60:12: fatal error: Arduino_Cellular.h: No such file or directory
     #include <Arduino_Cellular.h>
              ^~~~~~~~~~~~~~~~~~~~
  compilation terminated.
  
  Used library              Version Path
  ArduinoIoTCloud           2.0.4   /home/runner/Arduino/libraries/ArduinoIoTCloud
  Arduino_ConnectionHandler 0.9.0   /home/runner/Arduino/libraries/Arduino_ConnectionHandler
  Arduino_DebugUtils        1.4.0   /home/runner/Arduino/libraries/Arduino_DebugUtils
  WiFi                      1.0     /home/runner/.arduino15/packages/arduino/hardware/mbed_portenta/4.1.5/libraries/WiFi
  SocketWrapper             1.0     /home/runner/.arduino15/packages/arduino/hardware/mbed_portenta/4.1.5/libraries/SocketWrapper
  Ethernet                  1.0.0   /home/runner/.arduino15/packages/arduino/hardware/mbed_portenta/4.1.5/libraries/Ethernet
  GSM                       1.0     /home/runner/.arduino15/packages/arduino/hardware/mbed_portenta/4.1.5/libraries/GSM
  
  Used platform         Version Path
  arduino:mbed_portenta 4.1.5   /home/runner/.arduino15/packages/arduino/hardware/mbed_portenta/4.1.5
  Error during build: exit status 1
```

#  Portenta_BLE_Bridge.ino
```
Compiling sketch: Arduino_BHY2Host/examples/Portenta_BLE_Bridge
  In file included from /home/runner/Arduino/libraries/ArduinoIoTCloud/src/ArduinoIoTCloud.h:27:0,
                   from /home/runner/work/nicla-sense-me-fw/nicla-sense-me-fw/Arduino_BHY2Host/examples/Portenta_BLE_Bridge/Portenta_BLE_Bridge.ino:13:
  /home/runner/Arduino/libraries/Arduino_ConnectionHandler/src/Arduino_ConnectionHandler.h:60:12: fatal error: Arduino_Cellular.h: No such file or directory
     #include <Arduino_Cellular.h>
              ^~~~~~~~~~~~~~~~~~~~
  compilation terminated.
  
  Used library              Version Path
  ArduinoIoTCloud           2.0.4   /home/runner/Arduino/libraries/ArduinoIoTCloud
  Arduino_ConnectionHandler 0.9.0   /home/runner/Arduino/libraries/Arduino_ConnectionHandler
  Arduino_DebugUtils        1.4.0   /home/runner/Arduino/libraries/Arduino_DebugUtils
  WiFi                      1.0     /home/runner/.arduino15/packages/arduino/hardware/mbed_portenta/4.1.5/libraries/WiFi
  SocketWrapper             1.0     /home/runner/.arduino15/packages/arduino/hardware/mbed_portenta/4.1.5/libraries/SocketWrapper
  Ethernet                  1.0.0   /home/runner/.arduino15/packages/arduino/hardware/mbed_portenta/4.1.5/libraries/Ethernet
  GSM                       1.0     /home/runner/.arduino15/packages/arduino/hardware/mbed_portenta/4.1.5/libraries/GSM
  
  Used platform         Version Path
  arduino:mbed_portenta 4.1.5   /home/runner/.arduino15/packages/arduino/hardware/mbed_portenta/4.1.5
  Error during build: exit status 1
```

This occurs, even when Arduino_Cellular and ArduinoCellular are both added to the `depends` field of library.properties. Note that when compiler locally, the process is successful.

```
PS C:\GitHub\Arduino\nicla-sense-me-fw> arduino-cli compile -b arduino:mbed_portenta:envie_m7 .\Arduino_BHY2Host\examples\Nicla_IoT_Bridge\ --library .\Arduino_BHY2Host\
WARNING: library Arduino_BHY2Host claims to run on samd, mbed architecture(s) and may be incompatible with your current board which runs on mbed_portenta architecture(s).

Used library              Version Path
ArduinoIoTCloud           2.0.4   C:\Users\Ali Jahangiri\Documents\Arduino\libraries\ArduinoIoTCloud
Arduino_ConnectionHandler 0.9.0   C:\Users\Ali Jahangiri\Documents\Arduino\libraries\Arduino_ConnectionHandler
Arduino_DebugUtils        1.4.0   C:\Users\Ali Jahangiri\Documents\Arduino\libraries\Arduino_DebugUtils
WiFi                      1.0     C:\Users\Ali Jahangiri\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\4.1.5\libraries\WiFi
SocketWrapper             1.0     C:\Users\Ali Jahangiri\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\4.1.5\libraries\SocketWrapper
Ethernet                  1.0.0   C:\Users\Ali Jahangiri\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\4.1.5\libraries\Ethernet
GSM                       1.0     C:\Users\Ali Jahangiri\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\4.1.5\libraries\GSM
Arduino_Cellular          1.2.1   C:\Users\Ali Jahangiri\Documents\Arduino\libraries\Arduino_Cellular
StreamDebugger            1.0.1   C:\Users\Ali Jahangiri\Documents\Arduino\libraries\StreamDebugger
TinyGSM                   0.12.0  C:\Users\Ali Jahangiri\Documents\Arduino\libraries\TinyGSM
ArduinoHttpClient         0.6.1   C:\Users\Ali Jahangiri\Documents\Arduino\libraries\ArduinoHttpClient
ArduinoMqttClient         0.1.8   C:\Users\Ali Jahangiri\Documents\Arduino\libraries\ArduinoMqttClient
[... cut ...]
Arduino_BHY2Host          1.0.9   C:\GitHub\Arduino\nicla-sense-me-fw\Arduino_BHY2Host
ArduinoBLE                1.3.6   C:\Users\Ali Jahangiri\Documents\Arduino\libraries\ArduinoBLE
STM32H747_System          1.0     C:\Users\Ali Jahangiri\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\4.1.5\libraries\STM32H747_System

Used platform         Version Path
arduino:mbed_portenta 4.1.5   C:\Users\Ali Jahangiri\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\4.1.5
```

per1234, 2024-07-31:
@aliphys you can fix the failure by configuring the workflow to install the "Arduino_Cellular" library dependency, just as it does for other library dependencies.

Alternatively, the library could be added to the `depends` field of the "Arduino_ConnectionHandler" library's `library.properties` file, which will cause the library to be installed along with "Arduino_ConnectionHandler". That should probably be done anyway, but it does mean that the fix won't propagate to this repository until after a new release of "Arduino_ConnectionHandler" is made.

aliphys, 2024-08-01:
Workflow configured to include the library manually for the `Nicla_IoT_Bridge.ino` and `Portenta_BLE_Bridge.ino` sketches in https://github.com/arduino/nicla-sense-me-fw/pull/142/commits/db8303dc9322a577d78f808776e649ee6916630e.

aliphys, 2024-08-01:
Resolved in https://github.com/arduino/nicla-sense-me-fw/pull/142

closed 2024-08-01 by aliphys